awai54st/PYNQ-Classification #1: building error

wuweitao, 2017-09-14:
I followed the instructions form "https://github.com/awai54st/PYNQ-Classification", but encountered the following error when i tried "make all" after copying the makefile.config from PYNQ to caffe root. 

Iv tried method like adding opencv_imgcodecs to the LIBRARIES + line in the Makefile, but still, the problem remained. 

What am I do wrong?

CXX src/caffe/solvers/sgd_solver.cpp
CXX src/caffe/solvers/adagrad_solver.cpp
AR -o .build_release/lib/libcaffe.a
LD -o .build_release/lib/libcaffe.so.1.0.0
/usr/bin/ld: cannot find -lopencv_imgcodecs
/usr/bin/ld: cannot find -lopencv_imgcodecs
/usr/bin/ld: cannot find -lboost_python3
/usr/bin/ld: cannot find -lopenblas
collect2: error: ld returned 1 exit status
Makefile:572: recipe for target '.build_release/lib/libcaffe.so.1.0.0' failed
make: *** [.build_release/lib/libcaffe.so.1.0.0] Error 1

awai54st, 2017-10-08:
Sorry for the late reply. I have uploaded the SD card image onto an online drive. It has Caffe and Theano pre-installed. Please see the updated README.md for details.

awai54st, 2017-10-08:
And based on the log it looks like you are missing opencv and a boost (either openblas or atlas should do). If you wish to continue to install the dependencies manually, these may be the directions to start with.

closed 2017-10-08 by awai54st