awai54st/PYNQ-Classification #2: Caffe import issue on PYNQ

bharatsshinde, 2017-09-26:
When I try to import caffe on the PYNQ board I get an error. Do we need to make any changes to the caffe makefile before compiling it on the board?

awai54st, 2017-10-08:
Sorry for the late reply. I have uploaded  the SD card image onto an online drive. It has Caffe and Theano pre-installed. Please see the updated README.md for details.

closed 2017-10-08 by awai54st

qywan, 2017-12-27:
Hello, I want to know ，when  you do block design in Vivado, why simple_sum_0 and stream_mult_0 and mult_constant_0 IPs are necessary? Or  What's the use of the three IPs in block design?

awai54st, 2017-12-27:
Hi, these IPs are actually residues from a previous project. It should be okay to remove them. Sorry for the confusion.