awai54st/PYNQ-Classification #4: FPGA weight download

algoton, 2018-01-16:
First it is very creative project and I appreciate your effort to make the source open.
I have one puzzle. It appears you have converted Caffe model to Lasagne network first, then copy Caffe parameters to Lasagne network, then read out the paramters of Lasagne network before forwarding the weight parameters to FPGA. i.e. FPGALoadW(weight, 1, 32, 32, 2). My question is: Why not download the weight parameters of caffe model to FPGA directly? Thanks. Sean

weiwenjiang, 2018-01-20:
Yep, I have a similar question. 
I check your report and find that it claims that the added PYTHON layer in caffe may cause some degree of overhead on performance. And therefore, THEANO is adopted.
Could you please send the example of "fpga_conv_im2col.py" to me (jiang.wwen@pitt.edu)? I'd like to have a try to directly use caffe. Thanks!

awai54st, 2018-01-20:
Hi, the reason why I imported both caffe and theano is simply because I want to show that the framework supports data io in both caffe and theano formats. Actually the given notebook downloads weight directly from caffe pretrained model. You should be able to easily edit the notebook to import data in theano format as well.

closed 2018-01-20 by awai54st