awai54st/PYNQ-Classification #5: Performance of Lenet

weiwenjiang, 2018-02-05:
Hi, I've implemented the Lenet on PYNQ.
I found that the elapsed time for 500 images with the provided bitstream is roughly 1 second.
But if I used the bitstream generated by HLS and VIVADO tools with the provided projects, the elapsed time has increased to 5 seconds.
Could you please have a try and give some help on this?

awai54st, 2018-02-05:
Hi, 
Can I ask how did you test the LeNet function? Did you use the theano-packaged function? Or did you use the quick test function? The theano-packaged function requires some working - the first time you run it it will always give a very high latency, but the second time onwards it will show the correct speed. The quick test function works more reliably, and this problem should not occur.

Did you try with the bitstream that I provided? How was the performance?

If the provided bitstream works correctly, then it must be something to do with your vivado project. In that case I will need more information on how you regenerated the vivado project.

Best,
awai54st

weiwenjiang, 2018-02-05:
Hi,
I have tried the bitstream provided by you. The performance seems good.
And I only replace the bitstream generated by myself, based on the codes provided in this project, and it results in 5X degradation on performance.

I found that the Folding Factor in the hls source code is not consistent with your report. I just modified these factors, but the performance is not improved yet.

As for the generation of vivado project, I used the code provided by you.
(1) First, vivado_hls -f run_cifar_10.tcl
(2) Second, vivado_hls -f run_lenet.tcl
(3) Third, vivado -source pynq_arch.tcl
At this stage, the vivado project for Cifar is generated.
(4) Fourth, I replace the IP core of cifar by lenet obtained in the second step.
(5) Finally, bitstream is generated.

Is there anything wrong?

Thanks for your reply.

weiwenjiang, 2018-02-05:
PS.
I use both theano-packaged function and the quick test function. The results are the same.
I guess that there must be something wrong in generating the bitstream. 
Do you have any comments?

Best,
Weiwen

awai54st, 2018-02-05:
The design flow looks fine. Can I ask what do you mean by “folding factor not consistent to the report”?

The folding factor controls the parallelism of convolution. A higher folding factor gives better performance at the cost of more resource usage.

What folding factors did you use?

weiwenjiang, 2018-02-05:
I found in your report, the folding factor for "conv1", "conv2", "FC1", "FC2" are 25, 20, 25 and 20.
But in the hls file, these figures are 25, 25, 25, 10.
I've modified these factors, but it also does not work.

Best,
Weiwen

awai54st, 2018-02-05:
Did the same problem occur with Cifar10 example? Did you try the graphical design interface?

weiwenjiang, 2018-02-05:
I'm generating the bitstream of Cirfar10.
It will take a while.

BTW,
I'm using Vivado 2017.4. Will different versions of Vivado may lead to the difference?

awai54st, 2018-02-05:
It’s possible. I have never tried with 2017, and I believe there are significant behavioral differences with 2016. I have only tried with 2016.1, and based on feedbacks the implementation is stable with 2016.1.

Can you try with 2016.1 and let me know the results?

weiwenjiang, 2018-02-05:
The result of cirfar created by myself (vivado 2017.4) is the same with the one provided by you.

Can you send your lenet project (both hls and vivado) to me (jiang.wwen@pitt.edu)? And then I can have a try.
I think there may be something inconsistent between the project that you used to generate the bitstream and the one on github.

Best,
Weiwen

awai54st, 2018-02-05:
I have just regenerated the lenet example using a clean download from github, and did not encounter the problem.

The compiled project is 382.8MB. Are you sure that your email can receive such a large file?

I strongly suggest that you regenerate the lenet project using 2016.1

weiwenjiang, 2018-02-06:
The problem is solved by generating the bitstream in Vivado 2016.1.
It is hard to believe. Anyway, it works now.

Thanks for the help.

Best,
Weiwen

closed 2018-02-06 by awai54st