awai54st/PYNQ-Classification #12: output feature map

DIAMONDWHILE, 2018-03-21:
Excuse me
What should I do if I want to output 13 * 13 * 125 in FPGAQuickTest ?

awai54st, 2018-03-21:
Hi,
First you need to use the provided HLS project to synthesise a FPGA IP that does 13x13x125 output.
Once the bitstream is uploaded to the PYNQ board (in bitstream folder in the project), edit the parameters in the FPGAQuickTest function as follow:

`input_val = np.append([0, batch_size, KerDim, IFMCH, IFMDim, OFMCH, OFMDim, PadDim], input_val)`

Where you want OFMDim to be 13 and OFMCH to be 125.
I will re-upload my master thesis to the project, which contains the details of how the project has been designed.

Best,
awai54st

stanely0714, 2018-03-22:
oh~ I see
thanks for the help

closed 2018-05-07 by awai54st