awai54st/PYNQ-Classification #13: Query regarding the graphical design interface

arsam96, 2018-04-13:
I have flashed my sd card with the provided image and followed the script design interface steps for the the vivado. Now I want to know that the Graphical design interface says to use "make compile_graphical" command but where to use it? secondly the pdf report for the project says to open the CNN_BLOCK_DESIGN project in vivado but I can't find any file with such name.

awai54st, 2018-04-16:
Hi, 

For the first question, in a terminal go to `hw/graphical_design_flow/CIFAR-10`. Then `make compile_graphical` for compiling a CIFAR-10 example.

For the second question, the report was referring to an older version. "CNN_BLOCK_DESIGN" has now been replaced with a "sketchpad" project which will be auto-generated when you run `make compile_graphical.

Arsam1996, 2018-04-16:
Do I need to use the vivado command prompt for the make compile-graphical?

arsam96, 2018-04-16:
What is the next step step after implementing this command? How do I upload this on my board?

arsam96, 2018-04-16:
I am getting this error. Kindly help me out with this.
[image]

awai54st, 2018-04-16:
Hi,
Sorry this is an error in the very last line of sketchpad.tcl -- it is using the absolute path which is not applicable to your file system. I believe that the sketchpad project has been successfully created in your PC. The only thing you need to do next is to package this project into an IP block -- go to tools -> create and package IP. This IP block is the logic for a CIFAR-10 CNN model.

awai54st, 2018-04-16:
Yes. Up to this stage the HLS only generates the RTL codes -- which is whatever that have been packaged in the "sketchpad" IP. For bitstream, you need to go to the base project, which is a Vivado project, and import the "sketchpad" IP into it, then synthesise the base project to get a bitstream.

arsam96, 2018-04-16:
How can we import the sketchpad IP to the vivado project?

awai54st, 2018-04-17:
In base project, on its block diagram, right click, then add ip? The base project should’ve automatically linked the repo containing sketchpad ip block

arsam96, 2018-04-18:
I am really thankful to you in helping me out with the implementation of CNN. I have got a question that   how can we modify this so that it can identify images taken from a usb camera attached to the pynq?

arsam96, 2018-04-18:
I want to identify a number plate image taken from my usb camera. Kindly guide me with that.
Thanks.

awai54st, 2018-04-18:
There is one open source project, FINN by Xilinx, that did something very similar to what you described. 

https://github.com/Xilinx/BNN-PYNQ/blob/master/notebooks/BNN-from-webcam.ipynb

awai54st, 2018-05-06:
Guide here:
https://github.com/awai54st/PYNQ-Classification/tree/master/hw
Please be more specific on what step do you need help with.

arsam96, 2018-05-06:
Hey Erwei!

Just wanted to know If you have used the QNN repository for real time
object detection on pynq? I basically want to use the yolo script for real
time video detection from my usb camera. Can you help me with the code
please?

https://github.com/Xilinx/QNN-MO-PYNQ/tree/master/notebooks

Thanks.

awai54st, 2018-05-07:
Hi,
I haven't tried it yet, so unfortunately I cannot provide any guide on how to use that framework. However, I suggest that you have a look at another Xilinx framework named BNN on PYNQ (or FINN). That one also has a real time usb camera-based detection example. Both FINN and QNN are created mainly by the same author.
Please try to leave an issue on the QNN repo.

wangenyi, 2018-05-15:
hi 
  in script_design_flow,i can find axi_dma_slave/master and they connect the first  SCIG layer and the last FC layer  ,but in graphic_design_flow,i cant find them.why ?thank you!

wangenyi, 2018-05-15:
Erwei
   thanks for your answer, i can achieve the same design without  AXI_DMA_SLAVE/MASTER,right? And in 
script_design_flow ,i didnt find the testbench files. Without those files, can the program be carried out correctly? thank you very much!

awai54st, 2018-05-15:
Yes, and yes. The testbench is for simulation and debug only, not for hardware synthesis.

closed 2018-06-05 by awai54st