awai54st/PYNQ-Classification #15: Questions about the design process of the whole work.

bramblewalls, 2018-04-18:
Hi, i am a new developer in PYNQ, and i am considering deploy a cnn in pynq. Your work is really impressive to me. But i am sitll confused about the design process of the whole work. In other word, what should i do if i want to deploy my own cnn in pynq.Could you please give me some advices on the precess from traing a net to deploy a net in pynq.It would be very nice of you to give me these advices.I am looking forword to your reply.THX!

awai54st, 2018-04-18:
Hi,

Here are some possible steps:

Assuming you have finished designing and training your CNN, you should have the model (# of layers, size of weight matrices, word-width of the fixed-point parameters, etc) as well as the trained parameters for this model. Then, you can use the HLS part of this project to synthesise the RTL of your model, by selecting the appropriate hyper-parameters inside the C++ code of the HLS project. It is important to note that this project specialises in delivering small CNNs at high performance -- large CNN models may not fit, given the limited resources available on PYNQ-Z1. (For debugging, having C++ testbenches for HLS project is always useful.)

When HLS synthesis is complete, then you can import your output RTL into the Vivado part of this project ("base_project") for bitstream synthesis.

Then, connect to PYNQ using Jupyter notebook, and modify the hyper-parameters inside the python code (pay special attention to the "FPGA_QUICK_TEST" function) to accommodate your CNN model.

Then, load the bitstream that you have previously synthesised onto PYNQ using Jupyter notebook, execute the python code on PYNQ, and fingers crossed.

bramblewalls, 2018-04-27:
Thank you for your answer! I am now reading your technical report and tring to understand your hls code. There are same questions.first,i can see the function SCIG is for sliding window, and you say that "im2col" expands the input feature into column vectors. I am wondering that if you tranform a matirx into a vector and then how could you doing convolution on it? Second,it says in order to reduce buffer usage, you interleave the weights off-line. I think it should be important in your project and could you please make it more clearly? I guess these two question can be a bit complicated, it may take you long to give me a answer, so i think a blog or a paper about it will be ok for me. Sorry for bothering you and i am looking forward to your reply.

awai54st, 2018-04-30:
Hi,
By doing im2col you essentially unroll and expand the matrix, such that convolution becomes vector dot-products. Detailed explanation can be found in the Stanford CS231n notes. http://cs231n.github.io/convolutional-networks/
A more detailed explanation for interleaving can be found in Xilinx's FINN paper. https://arxiv.org/pdf/1612.07119.pdf

closed 2018-05-07 by awai54st

sabhizakaria, 2020-06-25:
Hi, first thank you for the wonderfull work, I m trying to implement a portion of the lenet5 ( basicaly conv and pool layers ) into fpga, and having read your user guide I have some issues finding the ip library that you used, should i synthesis each layer on its own and then export the output RTL into Vivado ?

awai54st, 2020-06-26:
Hi, assuming that you are using the script-based design flow, then you can find the ips here `PYNQ-Classification/hw/script_design_flow/hw_library/`

sabhizakaria, 2020-06-30:
Thanks for the reply, 

I tried to follow the instructions to implement cifar10 (or lenet5) using your framwork I got the following critical warning: 
 
[image]

So I tried to map the register automatically, but when executing the code in jupyter I get the following error : 

[image]

Any ideas about how to resolve it ? 

P.S : I m using the latest version of vivado suite 

Thank you

awai54st, 2020-07-01:
_So I tried to map the register automatically_
Yes, this should solve the problem.

_but when executing the code in jupyter I get the following error :_
The python script is looking for a StreamingSwitch hardware module name "axis_switch_0" which was instantiated in my origional block design. The purpose of that switch box was to have multiple functionalities on board which may or may not be relevant in your application.

sabhizakaria, 2020-07-01:
for now I just wanna replicate your implemtation on the PYNQ Z2 while following your steps and I still get the same keyError above when i try to call fpgaload function : 
 
[image]

awai54st, 2020-07-01:
This is now a different error right? Previously it was `KeyError: "SEG_axis_switch_0_Reg" ` but now it is `KeyError: "SEG_axi_dma_0_Reg"`. So now the script cannot find AXI DMA module. My suspicion is that the names of DMAs no longer match. When you port the project to Z2.

sabhizakaria, 2020-07-01:
Yeah due to some new standars the PYNQ-Z2 needs to follow some rules when namings dma ,but i thought  the pyhton script takes this change in consideration, I will try to see if there is a possibility to make it maches.

thanks

sabhizakaria, 2020-07-02:
hello again ,I ve tried to resynthesise the project using the
appropriate board  files of the Z2 ( from the xilinx web site), and yet i
still have the same problem
[image: image.png]

knowing that the modul axis DMA is properly set :
[image: image.png]

Any leads to how to adapt it so it could work ( will using the latest
version of vivado (2019) create such problems ? )

thank you

awai54st, 2020-07-02:
Hi I cannot see the attached error messages. I can only see [image: image.png]

sabhizakaria, 2020-07-03:
ah i think github didn't uploaded quite my images , as a i said i still have the same problem knowing that I ve switch the to the appropriate board files of the Z2 : 
[image]

[image]

awai54st, 2020-07-03:
What the python script is doing is that it searches for keyword "SEG_axi_dma_0_Reg" from the TCL file that was generated together with the bitstream. Have you updated the TCL file when you ported to PYNQ-Z2? If so, my suggestion is to look through the TCL file and see if you can find this keyword.

sabhizakaria, 2020-07-03:
Yeah i did and for some reason and as shown in the linked file i can neither  find the  "SEG_axi_dma_0_Reg" nor the "SEG_axis_switch_0_Reg" keword in the TCL file generated :

[decorator_cifar10_16b.txt](https://github.com/awai54st/PYNQ-Classification/files/4869585/decorator_cifar10_16b.txt)

sabhizakaria, 2020-07-06:
After comparing it with your tcl file ( the one in the sd image ) i found out that it replaced all the keywords ( registers )  with the option "-force" for some reason and i can't figure out why

awai54st, 2020-07-06:
Not sure. My tcl files were auto-generated by older versions of Vivado (before 2017.1), so it could be possible that more recent versions of Vivado have changed their rule when generating tcl files.

sabhizakaria, 2020-07-06:
Yeah it seems that after going back to the 2016.1 version the errors that I used to have are gone, 
Vivado tries to limits the portability of projects between that are implemented in different versions, which sometimes creates some problems

sabhizakaria, 2020-07-08:
Hey, it seems I have dma issues whenever i try to change the implementation of the architecture of the CNN (basically I want to use both fpga framwork and the cpu on to execute the cnn ) :   
[image]

[image]

awai54st, 2020-07-09:
The DMA is waiting for the FPGA module to return a "done" signal, but until time out it still didn't get that done signal. That's why the DMA will throw this error.

I suggest that you check if the python gives the correct number of input data to the FPGA, and if the FPGA is expecting the correct number of input data. If the FPGA module did not receive the correct number of inputs it will just keep waiting.