awai54st/PYNQ-Classification #18: Question on Jupyter

Yidagithub1103, 2018-05-03:
Hello!
I am a high school student from China.
I download the image you provided from Baidu Drive and use Win32DiskImager to put the image to sd card.
But i found an issue,I can't open the Jupyter Notebook from the static IP 192.168.2.99.
I have set up my static IP correctly.

awai54st, 2018-05-06:
Hi, 
For connection you may need to make sure that your terminal is under the same LAN address as the PYNQ board. If you are using home routers, normally 192.168.x.x is okay because that's a common default setting. However, if you are using public networks then you need to make sure that the first 3 numbers of your PYNQ's IP address match your computer's IP (which is assigned by the router, with dynamic IP assignment).
This may or may not be the case for you, but it's definitely worth checking. You can also try directly connecting the PYNQ board via micro usb (and putty), to make sure that the board is running.

Yidagithub1103, 2018-05-07:
Hi!
Connection part is definitely right,i use the office image and it runs correctly.
I mean does it have some issues when i use the image you provide?

awai54st, 2018-05-07:
I haven't encountered any issue like that before. Please try to use micro usb port for serial connection to the pynq board. Guide [here](http://pynq.readthedocs.io/en/latest/getting_started.html) Look for the "Opening a USB Serial Terminal" section. If the board does not respond to that then let me know.

If the 4 blue LED lights flash during the booting, then it suggests that the board has successfully booted from the image.

Yidagithub1103, 2018-05-08:
I  used micro usb port for serial connection to the pynq board,and the 4 blue LED lights successfully flashed,but i can not open the Jupyter still.

awai54st, 2018-05-09:
Then the image is running correctly on the pynq board. I think it is because of the IP address of the LAN connection. On the Internet there are numerous guides on LAN connection settings for Raspberry Pis, which are also relevant for configuring your pynq. Have a look at those. And also have a look at your local router's settings, and see what is required to set up a LAN connection in your local router.

closed 2018-06-05 by awai54st