awai54st/PYNQ-Classification #19: fixed_point_stream_convolution

DIAMONDWHILE, 2018-05-07:
HI~
I want to look at the output characteristics of each layer, so I split the model.I found that the output feature map of the first convolutional layer is different from the first layer output performed by ARM.So I look at the fixed_point_stream_convolution.h file.In this file, why not multiply the input picture?

awai54st, 2018-05-07:
Please specify: which part are you referring to? multiply with what?

DIAMONDWHILE, 2018-05-07:
I use script_design_flow
Convolution in fixed_point_stream_convolution.h
Why is the output feature map of the first CONV1 output different from the CAFFE output feature map?

awai54st, 2018-05-07:
For each layer, I scaled the weights and inputs with a scaling factor (to make them span the entire 8/16 bit signed int range) before casting them to fixed point for multiplication, such that the quantisation error is reduced (a somewhat inefficient implementation of "dynamic fixed point" quantisation). Hence, the output of each layer should also be scaled up as compared to the ARM benchmark, which uses floating point numbers.

So, ignoring the quantisation loss, the two output should also differ by a scaling factor.

DIAMONDWHILE, 2018-05-07:
In HLS, my first layer of convolution is
AXI_DMA_SLAVE(in_stream, connect_0);
SCIG<3, 3, 416, 4, 416, 1> (connect_0, connect_1);
SMM<1, 27, 4>(connect_1, connect_2, 1, 0, 25);
AXI_DMA_MASTER(connect_2, out_stream);
And CAFFE's
Feat = net.blobs['conv1'].data[0, :4]
Vis_square(feat)
The output matrix gap is too large

awai54st, 2018-05-07:
Your unrolling factor is 25 but the 2nd argument of SMM is 3x3x3=27. Vivado HLS expects the memory partitioning factor to be an integer factor of the memory size.

For debugging please write a HLS testbench about the first layer, and check the code’s correctness. The report provided in the repo should be enough to explain how the code works.

Shanyuyinglou, 2018-05-08:
@DIAMONDWHILE 
Hi,I am using windows10 and it supports vivado 2016.2 version.I just tried to install Vivado HLx 2016.2 version,but it  has no vivado command prompt.It just has vivado tcl shell and vivado hls command prompt.I tried 2017.2 and 2018.1 version and they also have no vivado command tool.I do not know how it happened. Is there any other way to run vivado's command line ?I have been looking for information on the Internet for a long time but I have not found any relevant information.Please give me some advise.
Thanks.
In addition,I am using vivado WebPACK which is the no cost.Which edition did you use,WebPACK ,design edition or system edition?

DIAMONDWHILE, 2018-05-09:
If I do not use "factors", how should I write
I want to confirm whether the reason for the failure is related to factors

awai54st, 2018-05-09:
The failure is probably not about the factors, but it's worth noting. You can try a factor like 3, 9, or 27 instead of 25. Regarding the source of failure, I really cannot tell based on the provided information. Please try to debug by using a testbench.

DIAMONDWHILE, 2018-05-10:
Could you teach me how to write a test bench?

awai54st, 2018-05-11:
Guide [here](https://medium.com/@chathura.abeyrathne.lk/xilinx-vivado-hls-beginners-tutorial-custom-ip-core-design-for-fpga-59876d5a4119)

DIAMONDWHILE, 2018-05-12:
I found that I changed the size of the image to 32 x 32 is the normal feature map
If the picture size is 416 x 416 it is not normal, which is why

DIAMONDWHILE, 2018-05-12:
The problem is solved
Thanks for your help

wangenyi, 2018-05-15:
hi 
  in script_design_flow,i can find axi_dma_slave/master and they connect the first  SCIG layer and the last FC layer  ,but in graphic_design_flow,i cant find them.why ?thank you!

awai54st, 2018-05-15:
Good to know! @DIAMONDWHILE

awai54st, 2018-05-15:
Answered [here](https://github.com/awai54st/PYNQ-Classification/issues/13#issuecomment-389147313).

Please do not duplicate your question on multiple issues.  @wangenyi

closed 2018-05-15 by awai54st