awai54st/PYNQ-Classification #20: Vivado command tool

Shanyuyinglou, 2018-05-07:
Hi,
I couldn't find out the vivado command prompt ,I just saw the vivado hls command prompt.So what exactly is the vivado command tool you were talking about?Thank you very much.

awai54st, 2018-05-07:
Are you using linux or windows? For windows there should be separate command tools for both vivado and vivado hls. Besides, I have seen reported issues saying that 2017 and 2018 versions do not work with this project. 2016.2 works most reliably.

Shanyuyinglou, 2018-05-08:
I am using windows10 and it supports  2016.2 version.I just tried to install 2016.2 version,but it still has no vivado command prompt.It just has vivado tcl shell and vivado hls command prompt.I do not know how it happened. Is there any other way to run vivado's command line ?I have been looking for information on the Internet for a long time but I have not found any relevant information.Please give me some advise.
Thanks.

awai54st, 2018-05-09:
You are right, the makefile is designed for linux terminals, not windows. Although with windows, you can always work around it, for example by executing each individual tcl script from the vivado_hls terminal.

The makefile itself is not important for regenerating the project; the tcl files called from inside this makefile are important.

Try to use the script interface instead -- that can be easier to use than the graphical version.

Shanyuyinglou, 2018-05-10:
I try to use vivado hls command prompt to run the first command in makefile as the screenshots.I intend to execute all the commands sequentially like this. Am I correct? If yes, what do I need to do next?
[image]
[image]

awai54st, 2018-05-11:
Yes what you are doing looks correct. Regarding the error, can I ask if you have copied the Arty-z7-20 board files to the Vivado installation? If not, guide [here](https://github.com/awai54st/PYNQ-Classification/tree/master/hw).

Shanyuyinglou, 2018-05-11:
Are you here now?About the setup of Theano on Python,I feel confused.I haven't see the related requirment in your meterial.Do I no longer need to install Theano? Is there any requirement for the format of SD card? I bought a 16G SD card myself. Is it because I format it and then copy the data to it? I'm confused about the initial setup steps for the board. Can you tell me in detail? I'm still a rookie so that have so many  issues, and I'd like to have your understanding for taking up your valuable time.

awai54st, 2018-05-11:
After you copy the board file, you need to restart Vivado for it to take effect. I haven't tried 2016.1. I have been using 2016.2 for developing this project. 

I have installed all the dependencies inside my sd card image, so no need for installing theano and so on. However, you should use softwares such as Win32DiskImager to burn the image into the SD card, not directly copy it to the card.

awai54st, 2018-05-11:
If Vivado still doesn't work, then please provide the synthesis log.

Shanyuyinglou, 2018-05-12:
**There is nothing in synthesis log as shown below.** 
[image]
**I can just post all the information from the TCL console window to you.Hope you can find clues.**
`start_gui
source G:/Vivado_HLS/2016.2/PYNQ-Classification-master/hw/graphic_design_flow/CIFAR-10/SKETCHPAD/sketchpad.tcl
# create_project sketchpad sketchpad -part xc7z020clg400-1
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository 'G:/Vivado/2016.2/data/ip'.
# set_property ip_repo_paths  .. [current_project]
# update_ip_catalog
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository 'c:/Users/hp/AppData/Roaming/Xilinx'.
WARNING: [IP_Flow 19-3656] If you move the project, the path for repository 'c:/Users/hp/AppData/Roaming/Xilinx' may become invalid. A better location for the repostory would be in a path adjacent to the project. (Current project location is 'c:/Users/hp/AppData/Roaming/Xilinx/Vivado/sketchpad'.)
# namespace eval _tcl {
# proc get_script_folder {} {
#    set script_path [file normalize [info script]]
#    set script_folder [file dirname $script_path]
#    return $script_folder
# }
# }
# variable script_folder
# set script_folder [_tcl::get_script_folder]
# set design_name sketchpad
# set run_remote_bd_flow 1
# if { $run_remote_bd_flow == 1 } {
#   set str_bd_folder base_project
#   set str_bd_filepath ${str_bd_folder}/${design_name}/${design_name}.bd
# 
#   # Check if remote design exists on disk
#   if { [file exists $str_bd_filepath ] == 1 } {
#      catch {common::send_msg_id "BD_TCL-110" "ERROR" "The remote BD file path <$str_bd_filepath> already exists!"}
#      common::send_msg_id "BD_TCL-008" "INFO" "To create a non-remote BD, change the variable <run_remote_bd_flow> to <0>."
#      common::send_msg_id "BD_TCL-009" "INFO" "Also make sure there is no design <$design_name> existing in your current project."
# 
#      return 1
#   }
# 
#   # Check if design exists in memory
#   set list_existing_designs [get_bd_designs -quiet $design_name]
#   if { $list_existing_designs ne "" } {
#      catch {common::send_msg_id "BD_TCL-111" "ERROR" "The design <$design_name> already exists in this project! Will not create the remote BD <$design_name> at the folder <$str_bd_folder>."}
# 
#      common::send_msg_id "BD_TCL-010" "INFO" "To create a non-remote BD, change the variable <run_remote_bd_flow> to <0> or please set a different value to variable <design_name>."
# 
#      return 1
#   }
# 
#   # Check if design exists on disk within project
#   set list_existing_designs [get_files */${design_name}.bd]
#   if { $list_existing_designs ne "" } {
#      catch {common::send_msg_id "BD_TCL-112" "ERROR" "The design <$design_name> already exists in this project at location:
#     $list_existing_designs"}
#      catch {common::send_msg_id "BD_TCL-113" "ERROR" "Will not create the remote BD <$design_name> at the folder <$str_bd_folder>."}
# 
#      common::send_msg_id "BD_TCL-011" "INFO" "To create a non-remote BD, change the variable <run_remote_bd_flow> to <0> or please set a different value to variable <design_name>."
# 
#      return 1
#   }
# 
#   # Now can create the remote BD
#   create_bd_design -dir $str_bd_folder $design_name
# } else {
# 
#   # Create regular design
#   if { [catch {create_bd_design $design_name} errmsg] } {
#      common::send_msg_id "BD_TCL-012" "INFO" "Please set a different value to variable <design_name>."
# 
#      return 1
#   }
[... 486 lines cut ...]
# preplace netloc axis_register_slice_7_M_AXIS 1 16 1 4160
# preplace netloc maxPool_CIF_0_1_0_out_r 1 5 1 1340
# preplace netloc axis_register_slice_1_M_AXIS 1 4 1 N
# preplace netloc axis_register_slice_0_M_AXIS 1 2 1 N
# preplace netloc axis_in_1 1 0 1 NJ
# preplace netloc FC_CIF_0_2_0_out_stream 1 21 1 N
# levelinfo -pg 1 0 180 440 710 980 1210 1440 1710 1970 2240 2510 2740 2980 3250 3510 3780 4060 4300 4540 4820 5100 5370 5550 -top 0 -bot 360
# ",
# }
# 
#   # Restore current instance
#   current_bd_instance $oldCurInst
# 
#   save_bd_design
# }
# create_root_design ""
ERROR: [BD 5-216] VLNV <xilinx.com:hls:FC_CIF_0_1:1.0> is not supported for the current part. 
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv xilinx.com:hls:FC_CIF_0_1:1.0 FC_CIF_0_1_0 "
    (procedure "create_root_design" line 41)
    invoked from within
"create_root_design """
    (file "G:/Vivado_HLS/2016.2/PYNQ-Classification-master/hw/graphic_design_flow/CIFAR-10/SKETCHPAD/sketchpad.tcl" line 313)`

Shanyuyinglou, 2018-05-12:
I found there is 'set_part {xc7z020clg**484**-1} -tool vivado'in fc_CIF_0_1/script.tcl , fc_CIF_0_2/script.tcl and so on but there is 'create_project sketchpad sketchpad -part xc7z020clg**400**-1'in SKETCHPAD/sketchpad.tcl.I guess this difference is the cause of error because Arty-z7-20 only has xc7z020clg**400**-1 part(xc7z020clg**484**-1 on ZYNQ-7) . So I changed '{xc7z020clg**484**-1' in every script.tcl into'{xc7z020clg**400**-1' and run it again.However,the same errors were still reported.Should I modify other files or commands？
PS：I very much hope to communicate with you in real time when you are free.If you can accept,take the liberty to ask you to add my WeChat(WeChat ID:山雨盈楼).I am a Chinese university student.If you feel inconvenience,please ignore my request.

Shanyuyinglou, 2018-05-12:
I tried to created a new project and run the tcl commands in sketchpad.tcl line-by-line in it to generate IP and connect ports,as shown below.
[image]
The following warning appeared.
[image]
But I don't understand the last two tcl commands:
**create_root_design ""
ipx::package_project -root_dir /home/awai54st/Documents/PYNQ-Classification/hw/graphic_design_flow/CIFAR-10/SKETCHPAD -vendor user.org -library user -taxonomy /UserIP -module sketchpad**
Then I skkipped these two commands and directly packaged the connected IP into a wrapper
The following warning appeared.
[image]
Then I synthesised and impletemented it successfully.But when I tried to write bitstream,it failed.
This is its bitstream log-------
_INFO: [Project 1-604] Checkpoint was created with Vivado v2016.2 (64-bit) build 1577090
open_checkpoint: Time (s): cpu = 00:00:31 ; elapsed = 00:00:42 . Memory (MB): peak = 709.523 ; gain = 502.109
Attempting to get a license for feature 'Implementation' and/or device 'xc7z020'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7z020'
Running DRC as a precondition to command write_bitstream
INFO: [DRC 23-27] Running DRC with 2 threads
ERROR: [DRC 23-20] Rule violation (NSTD-1) Unspecified I/O Standard - 72 out of 72 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: rsn_in, clk_in, axis_out_tdata[31:0], axis_out_tlast[0], axis_out_tready, axis_out_tvalid, axis_in_tdata[31:0], axis_in_tlast[0], axis_in_tready, axis_in_tvalid.
ERROR: [DRC 23-20] Rule violation (UCIO-1) Unconstrained Logical Port - 72 out of 72 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: rsn_in, clk_in, axis_out_tdata[31:0], axis_out_tlast[0], axis_out_tready, axis_out_tvalid, axis_in_tdata[31:0], axis_in_tlast[0], axis_in_tready, axis_in_tvalid.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_10_reg_3638_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_10_reg_3638_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_12_reg_3653_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_12_reg_3653_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_14_reg_3668_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_14_reg_3668_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_1_reg_3563_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_1_reg_3563_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_3_reg_3578_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_3_reg_3578_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_5_reg_3593_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_5_reg_3593_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_7_reg_3608_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_7_reg_3608_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
WARNING: [DRC 23-20] Rule violation (DPIP-1) Input pipelining - DSP sketchpad_i/FC_CIF_0_2_0/inst/r_V_9_reg_3623_reg input sketchpad_i/FC_CIF_0_2_0/inst/r_V_9_reg_3623_reg/B[17:0] is not pipelined. Pipelining DSP48 input will improve performance.
[... cut ...]
INFO: [Common 17-14] Message 'DRC 23-20' appears 100 times and further instances of the messages will be disabled. Use the Tcl command set_msg_config to change the current settings.
INFO: [Vivado 12-3199] DRC finished with 2 Errors, 104 Warnings
INFO: [Vivado 12-3200] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado 12-1345] Error(s) found during DRC. Bitgen not run.
INFO: [Common 17-83] Releasing license: Implementation
write_bitstream: Time (s): cpu = 00:00:28 ; elapsed = 00:00:16 . Memory (MB): peak = 845.047 ; gain = 135.523
ERROR: [Common 17-39] 'write_bitstream' failed due to earlier errors._

Shanyuyinglou, 2018-05-13:
Thanks to your help, I have successfully run on the board. If I want to export the data file of the experimental results (such as the classification results of the picture, accuracy, running time, etc.), can it be found in a certain path on the computer?

Shanyuyinglou, 2018-05-15:
I can't understand SDF and SDFG.SDF means Synchronous Dataflow.I learned it fom [https://blog.csdn.net/neustar1/article/details/20725777](url).But your pdf says,"My framework is based on the SDF paradigm. In this paradigm, a network is represented as a directed graph named SDFG. In SDFG, nodes represent computations and lines represent data streams. Fundamentally, the principle of SDF is that **whenever** input data are available for a node, the node will **immediately** start processing and generating outputs. When the entire network is constructed using SDF, each component IP on the graph can **independently** drive the data streaming, forming a heterogeneous streaming architecture. With streaming IO, output data immediately streams out **instead of being buffered in on-chip memory**, hence saving the memory footprint of the entire network."Can you explain it in detail?Thanks.

awai54st, 2018-05-15:
Related work [here](https://ieeexplore.ieee.org/abstract/document/7544745/).

closed 2018-06-05 by awai54st