awai54st/PYNQ-Classification #21: the PYNQ resources usage

sdnuzwk, 2018-05-27:
Dear prof,
  First,i am very lucky for studying your project and I have implemented the project on vivado_hls 2017.4.(cifar and lenet).
But it seems that the resources on PYNQ is not enough.The following is my results
[image]
[image]

sdnuzwk, 2018-05-27:
thank you sir

awai54st, 2018-05-27:
Hi, thank you.
As some previous issues have pointed out, this project works most stably on 2016.2. There have been reported issues with projects synthesised in 2017.1.
Can I ask which benchmark are you trying to implement? Is it lenet or cifar10, or is it your own CNN architecture?

sdnuzwk, 2018-05-28:
@awai54st 
Dear prof，
        I fell excited for receiving your reply，thank you.I copy your project code about lenet and cifar10 on vivado_hls 2017.4，i do not think the version of vivado_hls is able to lead the different systhesis results，my systhesis results about lenet and cifar10 on vivado_hls 2017.4 are showed in the previous question，could you give me some advice？I am so confused because i think i follow your steps completely.
Best wishes

sdnuzwk, 2018-05-29:
please help me sir ,i am in confused

sdnuzwk, 2018-05-30:
Dear Prof,
  Let me explain in detail the process of my experiment and show the comprehensive report.
  I implemented cifar10 routines on vivado_hls2017.4 and vivado_hls2016.2 respectively.
1.On vivado_hls2017.4
[image]
the systhesis results about cifar10 on vivado_hls 2017.4 are showed in the picture.As you can see,the using number of LUTs are 89096,and the Estimated time is 12.59ns.

The console reports are follow:
     Starting C synthesis ...
/opt/Xilinx/Vivado/2017.4/bin/vivado_hls /home/sdnuzwk/PYNQ-Classification-master/hw/script_design_flow/CIFAR_10_wrapper/CIFAR_10/solution1/csynth.tcl
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2017.4/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'sdnuzwk' on host 'node01' (Linux_x86_64 version 3.10.0-514.el7.x86_64) on Wed May 30 10:39:23 CST 2018
INFO: [HLS 200-10] On os "CentOS Linux release 7.3.1611 (Core) "
INFO: [HLS 200-10] In directory '/home/sdnuzwk/PYNQ-Classification-master/hw/script_design_flow/CIFAR_10_wrapper'
INFO: [HLS 200-10] Opening project '/home/sdnuzwk/PYNQ-Classification-master/hw/script_design_flow/CIFAR_10_wrapper/CIFAR_10'.
INFO: [HLS 200-10] Adding design file '../hw_library/stream_convolution_slideWindow.h' to the project
INFO: [HLS 200-10] Adding design file '../hw_library/pool.h' to the project
INFO: [HLS 200-10] Adding design file '../hw_library/fully_connected.h' to the project
INFO: [HLS 200-10] Adding design file '../hw_library/fixed_point_stream_convolution.h' to the project
INFO: [HLS 200-10] Adding design file 'config.h' to the project
INFO: [HLS 200-10] Adding design file '../hw_library/axi_dma_slave.h' to the project
INFO: [HLS 200-10] Adding design file '../hw_library/axi_dma_master.h' to the project
INFO: [HLS 200-10] Adding design file 'CIFAR_10_wrapper.h' to the project
INFO: [HLS 200-10] Adding design file 'CIFAR_10_wrapper.cpp' to the project
INFO: [HLS 200-10] Opening solution '/home/sdnuzwk/PYNQ-Classification-master/hw/script_design_flow/CIFAR_10_wrapper/CIFAR_10/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg484-1'
INFO: [HLS 200-10] Analyzing design file 'CIFAR_10_wrapper.cpp' ...
INFO: [HLS 200-10] Validating synthesis directives ...
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:00:48 ; elapsed = 00:00:19 . Memory (MB): peak = 449.988 ; gain = 13.379 ; free physical = 14405 ; free virtual = 124738
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:00:49 ; elapsed = 00:00:20 . Memory (MB): peak = 449.988 ; gain = 13.379 ; free physical = 14402 ; free virtual = 124740
INFO: [HLS 200-10] Starting code transformations ...
INFO: [HLS 200-111] Finished Standard Transforms Time (s): cpu = 00:00:51 ; elapsed = 00:00:22 . Memory (MB): peak = 450.359 ; gain = 13.750 ; free physical = 14380 ; free virtual = 124727
INFO: [HLS 200-10] Checking synthesizability ...
INFO: [HLS 200-111] Finished Checking Synthesizability Time (s): cpu = 00:00:52 ; elapsed = 00:00:23 . Memory (MB): peak = 577.984 ; gain = 141.375 ; free physical = 14365 ; free virtual = 124713
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'L3' (./../hw_library/fixed_point_stream_convolution.h:124) in function 'SMM<1u, 800u, 64u>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'L3' (./../hw_library/fixed_point_stream_convolution.h:124) in function 'SMM<1u, 800u, 32u>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'L3' (./../hw_library/fixed_point_stream_convolution.h:124) in function 'SMM<1u, 75u, 32u>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'Loop-1' (./../hw_library/stream_convolution_slideWindow.h:123) in function 'SCIG<5u, 3u, 32u, 32u, 32u, 2u>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'Loop-1' (./../hw_library/stream_convolution_slideWindow.h:123) in function 'SCIG<5u, 32u, 8u, 64u, 8u, 2u>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'Loop-1' (./../hw_library/stream_convolution_slideWindow.h:123) in function 'SCIG<5u, 32u, 16u, 32u, 16u, 2u>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'L3' (./../hw_library/fully_connected.h:124) in function 'FC<1u, 64u, 10u>' for pipelining.
INFO: [XFORM 203-502] Unrolling all sub-loops inside loop 'L3' (./../hw_library/fully_connected.h:124) in function 'FC<1u, 1024u, 64u>' for pipelining.
INFO: [XFORM 203-501] Unrolling loop 'Loop-1.1' (./../hw_library/pool.h:124) in function 'pool<2u, 64u, 8u>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Loop-2' (./../hw_library/pool.h:130) in function 'pool<2u, 64u, 8u>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Loop-1.1' (./../hw_library/pool.h:124) in function 'pool<2u, 32u, 32u>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Loop-2' (./../hw_library/pool.h:130) in function 'pool<2u, 32u, 32u>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Loop-1.1' (./../hw_library/pool.h:124) in function 'pool<2u, 32u, 16u>' completely.
INFO: [XFORM 203-501] Unrolling loop 'Loop-2' (./../hw_library/pool.h:130) in function 'pool<2u, 32u, 16u>' completely.
INFO: [XFORM 203-501] Unrolling loop 'L4' (./../hw_library/fixed_point_stream_convolution.h:126) in function 'SMM<1u, 800u, 64u>' completely.
INFO: [XFORM 203-501] Unrolling loop 'L4' (./../hw_library/fixed_point_stream_convolution.h:126) in function 'SMM<1u, 800u, 32u>' completely.
[... cut ...]
INFO: [VHDL 208-304] Generating VHDL RTL for cifar_10.
INFO: [VLOG 209-307] Generating Verilog RTL for cifar_10.
INFO: [HLS 200-112] Total elapsed time: 101.54 seconds; peak allocated memory: 449.345 MB.
Finished C synthesis.

2.On vivado_hls2016.2
[image]
the systhesis results about cifar10 on vivado_hls 2016.2 are showed in the picture.As you can see,the using number of LUTs are 77036,and the Estimated time is 11.37ns.
why have two different results?

sorry to disturb you,but i am in mess....

sdnuzwk, 2018-05-31:
OK，let me try again，sir

sdnuzwk, 2018-06-04:
Dear Prof，
 In the previous issues,you think the simple_sum_0 and stream_mult_0 and mult_constant_0 IPS are not necessory. 
 I want to know if i only delete the following IPS block design （simple_sum_0 and stream_mult_0 and mult_constant_0） in vivado,will this project still work?

sdnuzwk, 2018-06-04:
Dear Prof,
i have another question,that is the FPGA overall architecture in vivado,you can see in the picture
there are two IPS make me confuse,the one is in user ip sketchpad_v1_0,the other is in vivado_hls ip Cifar-10,so i want to know ,if i want to generate a correct bitstream,which ip i should choose?
Best wishes!
[image]

awai54st, 2018-06-04:
Yes, removing simple_sum_0 and stream_mult_0 and mult_constant_0 will not affect the project.

awai54st, 2018-06-04:
I think the photo you showed me is your Vivado IP repository?
I don't really know what you have synthesised so far, but all IPs generated from this project are stored in the same repository, which is the one that you are showing in this photo. So far I can see IPs from both the "graphical" and "script" design process.
In my instructions I have explained that sketchpad function is a vivado block design project for the "graphical" design process. I think you are trying to use the "script" design process instead? So that means the sketchpad function is irrelevant to you.
Please read the instructions carefully.

sdnuzwk, 2018-06-05:
Dear Prof,
 Thanks for your detail reply,thanks again.
 But i want to make sure my confusion.
Question 1: 
In the one hand,when i do make compile_graphical command in vivado,followed by generating the SKETCHPAD project,i think the SKETCHPAD is a CNN models for cifar-10,so i think the sketchpad_v1_0 in the UserIP is the cifar-10 IP,correct?
 In the other hand,in vivado_hls,i run the following command vivado_hls -f run_cifar_10.tcl,there also can produce an IP named cifar-10,so in the Vivado IP repository has two IPS make me confused,one is the sketchpad_v1_0,the other is cifar-10,so my question is if i want to verify the cifar-10 project，is the role of these two IPS the same？
Question 2:
If i want to design my own CNN architecture
My first idea:Should i only modify the codes in vivado_hls,then i synthesise the code in vivado_hls and export the rtl ,in the next step,then a IP core of my own CNN is obtained, i just need to add this IP to the overall architeture,is my idea correct?
My second idea:Should i chaining layer blocks to design my CNN models,then package it into an IP,then a IP core of my own CNN is also obtained.
which is the correct idea?
Thank you sir .
Best wishes!

awai54st, 2018-06-05:
Hi, 
Q1. Yes the two IPs should function equivalently. The SKETCHPAD function is a vivado function designed only to package layer IPs into a complete network block design. So for graphical process this is the network IP. For script process the networks will be automatically packaged in hls projects. SKETCHPAD project is not needed for script process.
Q2. Both ideas are correct. In fact idea 1 and 2 corresponds to script and graphical process, respectively.

sdnuzwk, 2018-06-06:
Thanks for your instructions very very much.Best best wishes!!

closed 2018-06-06 by sdnuzwk