awai54st/PYNQ-Classification #22: error about make compile_graphical

YzwWh9327, 2018-06-04:
When I use HlS Command prompt run"make compile_graphical",there are some errors.
[image]

awai54st, 2018-06-05:
HI,
From the log I can see that all HLS IPs have been synthesised, and the error appears in the vivado command `source run_ippack.tcl -notrace`. 
In the makefile, the only vivado command comes from building the "sketchpad" vivado project. However, in my provided tcl file this line of command does not exist. So my guessing is that an incorrect tcl file has been executed.
This appears to be because you used the Windows vivado_hls command prompt to execute a vivado command. In fact this makefile has been designed mainly for Linux OS. Changes might be needed when using it in a Windows environment. 
Can you check if there is a new folder named "build" under the "CIFAR-10" folder? If so, cd to build/SKETCHPAD, then execute `vivado -source sketchpad.tcl` in a Vivado command prompt (not a Vivado HLS command prompt).

closed 2018-06-12 by awai54st

shahmustafa, 2018-08-20:
while executing `vivado -source sketchpad.tcl`  in vivado command prompt, the Viivado GUI gets stated and showing these errors
[image]

awai54st, 2018-08-28:
I can see that you are using vivado 2018. I haven't tested this project on it and hence I cannot guarantee that there's no incompatibility issues.

By default the tcl file sketchpad.tcl includes ips using `set_property ip_repo_paths .. [current_project]`. It seems that this line of code has failed in your case. I suggest that you debug from this.