awai54st/PYNQ-Classification #24: the usage

sdnuzwk, 2018-06-07:
Dear Prof,
   Sorry to disturb you again!
   I  have a resource utilization issue that I want to discuss with you，because of the limitation of pynq board，if i want to design a deep net，the resource is not enough，so i  want to design a dropout layer after the pooling layer,my idea is multiply the output of the pooling layer by a constant less than 1，but  i don't know how to modify the code in the definition about the pooling layer ,could you give me some specific advices about the pooling code?

sdnuzwk, 2018-06-07:
Sorry,another question，i want to know how to make sure the second parameter(A_ROW_MAX) of SMM,how to compute it?

sdnuzwk, 2018-06-08:
Dear Prof,
   The FC layer is followed by the pool layer,how to make sure the second parameter of the FC layer,in your instructions,the second parametr is A_ROW_MAX=KER*KER*IN-CH。
   For example,in cifar 10,the code is 
        pool<2, 64, 8>(connect_8, connect_9, 3, 1, 0);
	FC<1, 1024, 64>(connect_9, connect_10, 4, 0, 32);
  so the 1024=4*4*64,so the KER is 4?Why?I am confused.i think it's 2

sdnuzwk, 2018-06-11:
@awai54st

awai54st, 2018-06-12:
Q1: Multiply your factor at line 172 of pool.h
Q2: I can't understand your question. For a FC layer, shouldn't the kernel size be 1?
Q3: The main.cpp is testbench, not the actual hardware. I used 28 for testing only. If you look at the CIFAR_10_wrapper.cpp (which is actually the hardware), then you can see that the input dimension is 32.

sdnuzwk, 2018-06-13:
First of all, thank you for the guidance you gave me and help me a lot.
  The Q2:
  I wan to know compute  the second parameter of the FC layer,in your instructions,the second parametr is equal to  A_ROW_MAX=KER*KER*IN-CH。
For example,in cifar 10,the code is
pool<2, 64, 8>(connect_8, connect_9, 3, 1, 0);
FC<1, 1024, 64>(connect_9, connect_10, 4, 0, 32);
so the 1024 is the second parameter,1024=4*4*64,64 is the IN-CH,so the KER is 4?

sdnuzwk, 2018-06-14:
[image]
Dear Prof,
     Is the matrix multiplication picture right?In ADDITION1,the A2*B2 is added double，isn't it？

awai54st, 2018-06-26:
In a fully connected layer, the operation is dot product between input feature map and weight matrix.
For your example, input frame has 4x4x64=1024 pixels. The corresponding weight matrix has 4x4x64x64 parameters, such that after dot product the output is in 1x1x64.

awai54st, 2018-06-26:
You are right, the A2xB2 should only be accumulated once. Thank you!

sdnuzwk, 2018-06-29:
@awai54st Thank you very much ,prof.

closed 2018-06-29 by sdnuzwk

sdnuzwk, 2018-09-02:
@awai54st

awai54st, 2018-09-07:
Hi, sorry for the late reply. Yes, I think you are right, they should have the same limit.

sdnuzwk, 2018-09-11:
@awai54st Dear Professor, thank you for your detailed answers. In Section 6.2, you designed a single convolutional layer and tested resource usage with 32-bit floating point numbers. Can you guide me through the specific method steps? How do I change the code?

awai54st, 2018-09-11:
You can look for the definition of AXI_VAL. Currently it is 8 or 16 bit fixed. Change it to float.  (PYNQ-Classification/hw/script_design_flow/LeNet_wrapper/config.h)

sdnuzwk, 2018-09-11:
@awai54st If I want to test with just one convolution layer, how do I need to modify it in the cpp file?

awai54st, 2018-09-11:
You can make a CNN model with one layer.

sdnuzwk, 2018-09-11:
@awai54st。Does a convolutional layer include the SGIG plus SMM functions or only the SGIG function?

awai54st, 2018-09-11:
SCIG+SMM

sdnuzwk, 2018-09-11:
@awai54st thank you very much, Rose

sdnuzwk, 2018-09-17:
@awai54st Dear Professor, I am sorry to bother you. I designed a network myself, but why is the number of LUTs used far higher than the 53200? It should be reasonable to say that the resource utilization of BRAM and DSP is not very high.

awai54st, 2018-09-17:
Hi, unfortunately I cannot help you with this. There are many possible reasons to that. I suggest that you have a look at the basics of low-level FPGA design.

sdnuzwk, 2018-09-20:
@awai54st thank you very much,sir