awai54st/PYNQ-Classification #30: about the conv_fpga.py file

liubinfeng, 2018-11-30:
hi，
     When I read your source file **conv_fpga.py**, I found that all the keys you used to index the IPs in _PL.ip_dict_  dictionary have a prefix ‘SEG’ and a suffix ‘Reg’. As follows：
    _class StreamingSwitch:
        def __init__(self,name):
            base_addr=int(PL.ip_dict["SEG_{0}_Reg".format(name)][0],16)
            self.mmio=MMIO(base_addr,256)
            self.reset()_
    I have printed the _PL.ip_dict.keys()_ and the output is:
        _dict_keys(['axi_dma_0', 'axi_dma_1', 'axis_switch_0', 'mult_constant_0'])_
    As you can see ,with no Prefix. So could you please explain why the prefix and suffix are added here?
Thanks a lot,
liu

awai54st, 2018-11-30:
Hi Liu, unfortunately I don't know the answer for that - this comes from an example design of an old version of pynq code from xilinx on github. The entire project was developed with that example as skeleton.

liubinfeng, 2018-12-03:
Hi Mr Wang,
Thanks for your help! I think I have found the reason. Maybe your work is based on pynq 1.4 or some other previous versions, and now the pynq source has been updated to version 2.3. The two versions of the source code are somewhat different, so it caused me confusion. Thank you ~

closed 2018-12-03 by liubinfeng