awai54st/PYNQ-Classification #31: SD CARD IMAGE

xiaozhupeiqi666, 2018-12-04:
Hi，
   I want to get a SD card image with pre-installed Caffe and Theano dependencies ,so I download your pynq-cnn-image file from Download Link in the README.md，but my pynq board is zcu104 and I am not sure if the image file can start my board or I mistake how to use the the image file?

awai54st, 2018-12-04:
Hi, unfortunately the SD card image was only designed for zynq xc7z020 (PYNQ-Z1). The project can work with zcu104 but extra modifications need to be made.

xiaozhupeiqi666, 2018-12-05:
Hi,as what you say,I tryed to put the downloaded image into my SD card as the start image of ZCU104 but failed.The project has been modified as my zcu104 project successfully,But caffe or theano dependencies really bothered me because I was a hardware developer.Now I wonder if [manual_instal. md] is the only way out for me, or if you can link the tutorial on making sd card images. I really appreciate your reply,thanks

awai54st, 2018-12-10:
Hi, unfortunately I did that a long time ago, with quite a lot of trial and errors, and couldn't find the step-wise tutorial on how to do this.
Mostly I installed the dependencies onto the Linux OS terminal of PYNQ as if it was a normal computer, except with an ARM chipset instead of Intel. 
Hope it was helpful

closed 2019-01-14 by awai54st