awai54st/PYNQ-Classification #32: Lenet-5 Block Design IP with base project

centrixsol, 2018-12-13:


centrixsol, 2018-12-13:
Hi there,
You have done a great project and made it open source, I have been working on recreating this project to make custom network. when I open your Cifar-1o block design project and convert it into lenet-5 by removing exta layers and genrate an IP block and then use this IP in base project. I face the following issues. Please help me on how I can design my own network similar to lenet-5. please view the screenshot of error.

I will wait eagerly for your reply.
[image]

awai54st, 2018-12-15:
Hi, sorry it is a mistake that I made in my code. Please go to the tcl file you used to generate the project, and replace all address pointing to /home/awai54st/etc with the address of your project files.
I strongly advice that you try the script-based flow instead, which is a lot more stable.

closed 2019-01-14 by awai54st