awai54st/PYNQ-Classification #39: FPGA Weights Copying Issue

centrixsol, 2019-01-15:
Hi Erwei Wang,
First of all Thank you so much for doing this cool project. I have trained my network using caffe's cifar10_full with 60,000 iterations. when I try to copy the paramters from caffe to lasagne i run into the following error 
Key error "ip2"
Can you please look at it and suggest why I am running into this problem. I have modeified Solver.prototxt and have changed ip2 layers number of outputs from 10 to 2.

net = {}
net['input'] = InputLayer((None, 3, 32, 32))
net['conv1'] = ConvLayer(net['input'], num_filters=32, filter_size=5, pad=2, nonlinearity=None)
net['pool1'] = PoolLayer(net['conv1'], pool_size=2, stride=2, mode='max', ignore_border=False)
net['relu1'] = NonlinearityLayer(net['pool1'], rectify)
net['conv2'] = ConvLayer(net['relu1'], num_filters=32, filter_size=5, pad=2, nonlinearity=rectify)
net['pool2'] = PoolLayer(net['conv2'], pool_size=2, stride=2, mode='average_exc_pad', ignore_border=False)
net['conv3'] = ConvLayer(net['pool2'], num_filters=64, filter_size=5, pad=2, nonlinearity=rectify)
net['pool3'] = PoolLayer(net['conv3'], pool_size=2, stride=2, mode='average_exc_pad', ignore_border=False)
net['ip1'] = DenseLayer(net['pool3'], num_units=64, nonlinearity = None)
net['ip2'] = DenseLayer(net['ip1'], num_units=2, nonlinearity = None)
net['prob'] = NonlinearityLayer(net['ip2'], softmax)
import numpy as np

layers_caffe = dict(zip(list(net_caffe._layer_names), net_caffe.layers))
for name, layer in net.items():
    try:   
        if name=='ip1'or name=='ip2':
            layer.W.set_value(np.transpose(layers_caffe[name].blobs[0].data))
            layer.b.set_value(layers_caffe[name].blobs[1].data)
        else:
            layer.W.set_value(layers_caffe[name].blobs[0].data[:,:,::-1,::-1])
            layer.b.set_value(layers_caffe[name].blobs[1].data)
            
    except AttributeError:
        continue
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-5-b759ff156a6b> in <module>()
      8     try:
      9         if name=='ip1'or name=='ip2':
---> 10             layer.W.set_value(np.transpose(layers_caffe[name].blobs[0].data))
     11             layer.b.set_value(layers_caffe[name].blobs[1].data)
     12         else:

KeyError: 'ip2'

awai54st, 2019-01-17:
Hi,
Have you updated CIFAR_10.prototxt and cifar10_quick_iter_5000.caffemodel.h5 accordingly? These files are the pretrained network and you need to replace them with yours.

centrixsol, 2019-01-18:
Yes, I have been able to resolve this issue. However I am unable to get my head arround how did you convert images data to numpy array. In CIFAR-10 Example.

data = np.load('cifar10.npz')
mean_image = np.load('mean_image.npy')
data_zeromean = data['raw'] - mean_image

def make_image(X):
    im = np.swapaxes(X.T, 0, 1)
    im = im - im.min()
    im = im * 1.0 / im.max()
    return im
plt.figure(figsize=(5, 5))
plt.imshow(make_image(data['raw'][0]), interpolation='nearest')

Since I have my own dataset with 2 classes. How Can I generate this .npz and .npy files. Do you have a script to generate these files for numpy.

Thank you so so much for your replies.

awai54st, 2019-01-18:
Unfortunately I do not have the script with me anymore. It's a fairly simple script though, since all it needs to do is to compress jpegs into a npz file. The mean image should be computed as the mean of your dataset.

They don't have to be npz and npy. Please feel free to edit the code to use formats of your own choosing.

closed 2019-06-18 by awai54st