awai54st/PYNQ-Classification #45: Modifications in the network

s95github, 2019-06-15:
Hi,
I have made some changes in the wrapper file and fixed_point_stream_convolution.cpp file in order to test the network with different kernel sizes. I have generated the hls ip and the Vivado design which allowed me to generate the bit and tcl files. Now I want to ckeck if my generated files are correct, could you please give me the steps from the beginning until obtaining the bit and tcl files, because I am afraid that i have missed some steps? 
and also how to put this files in the FPGA chip?
Thanks in advance.

awai54st, 2019-06-17:
Hi, it looks like you have successfully generated the bitstream and tcl files (judging from your description). 

To put the bitstream and tcl files to the chip you should go to pynq's jupyter notebook and link them in the python source code. Also, you need to change the python source code according to your architectural modification, so that the hardware (reconfigurable logic, programmed in your bitstream and tcl) and software (CPU program, programmed in python) are in sync with each other.

s95github, 2019-06-17:
These are all the changes that I have made:
- I modified the prototxt file for a kernel size of 1x1 in each convolution layer, with a pading=0.
- I generated a new caffemodel file corresponding to the prototxt file, the weights are random values, because I'm not looking for precision, I just want to check the time execution of the fpga deployment.
[image]

- I have generated the bitstream and tcl files by using Vivado (picture1)
[image]
 after making modifications in the CIFAR_10_wrapper.cpp and changed the factor in SMM file (picture 2,3)
[image]
[image]

- I have changed the name of the bitstream in conv_fpga.py (picture 4)
[image]

- I have made some changes in the python code source (picture 5,6,7)
[image]
[image]
[image]

- I have obtained an error while deploying the network in the FPGA (picture 8,9) 
[image]
[image]

I  couldn't understand the origin of the error. Can you help me please?

awai54st, 2019-06-18:
Basically this message means that the output data stream from your IP never had the "done" signal high.

There could be many reasons to that. Several possibilities are listed below:

1. The python code didn't stream enough input data to the IP so that it is still waiting for new input data.
2. The python code streamed correct number of inputs to the IP, but the IP was expecting too many input data.
3. Both.

If you look at the HLS source code, you can see that the "done" signal is instantiated as a "bool". A "done=True" marks the last data in the output data stream.

s95github, 2019-06-19:
Ok. I will check, thank you very much.

s95github, 2019-06-19:
Please, which HLS source code do you mean, can you give me its name?

s95github, 2019-06-19:
After checking, I have noticed that I did not made any change to pynq_arch.tcl (responsible for calling vivado and generating the bitstream). How can I tell it to use my IP with the block design (base project)?

It might be that the generated bitsream is using the old IP since I did not told it to specifically choose mine.

awai54st, 2019-06-20:
I think you do not need to modify the tcl file. Go to your tcl file and record the name of your ip (for example, the CIFAR-10 example's ip name is `xilinx.com:hls:cifar_10:1.0`).

Go to `fpga_conv.py` line 196 and update the ip name, so that the python code will go to your tcl file and execute the correct ip hardware.

closed 2019-09-20 by awai54st