awai54st/PYNQ-Classification #46: Changing HLS IP

Rana-Yehya, 2019-08-01:
Hi, 
Do I have to change the HLS structure an implementing a new IP to work with my CNN model?

awai54st, 2019-08-09:
Hi,  sorry for the late reply. This old issue might have the answer to your qn https://github.com/awai54st/PYNQ-Classification/issues/15.

Rana-Yehya, 2019-08-10:
Hi, I understand now the steps. Thank you. But my model has rectangle image size and its dimension is (height =114, width=59, channels=22). Should I enter the biggest size 114 into the ifsize in HLS code?

Rana-Yehya, 2019-08-10:
Hi, I understand now the steps. Thank you. But my model is taking the shape (1,22,59,114). Should I enter the biggest size in the "IFDim_curr" in HLS code?

awai54st, 2019-08-12:
Perhaps an easy way to get around that is to zero-pad the kernels/input feature maps into square shape, though it also leads to wasted computations. 

It's definitely doable to change the model size in the HLS code, but I think you might need to edit the source code by quite a bit. For example, the sliding window for convolution assumes a square-shaped input feature map, so that has to be edited. The "done" signal in the hand-shaking protocol also needs to be triggered at the correct timing (i.e. high at the very last pixel), otherwise the DMA will be stuck until timeout. These are two important things that I can think of at the moment.

closed 2019-09-20 by awai54st