awai54st/PYNQ-Classification #48: Constructing CNN

Elsaam2y, 2020-01-15:
Hallo, 
Thank you very much for this amazing work and for sharing it with others. I need only to ask about the construction of CNN steps as you have mentioned in your report on page 58. The first step was to open the CNN_Block_Design project, however, I can't find this file. I guess that I am missing something and I would be glad if you could explain it to me. 

Thanks in advance

awai54st, 2020-01-20:
Hi, thank you for your interest in my work. The instructions given in the report describes the "graphical design flow" which many people encountered difficulty while reimplementing. Therefore I made the "script design flow" which works more robustly for most people. `hw/README.md` contains instructions.

Elsaam2y, 2020-01-28:
Thanks for your response and for your hard work. Sorry for my late response.

closed 2020-02-03 by awai54st