awai54st/PYNQ-Classification #49: Can you share a make file for lenet5 architecture

nehru98, 2020-03-02:
Ive tried regenerating . Tcl files from lenet5 directories, however its not showinf the design and shows an ip error... It would be useful if we could get a make file for lenet5 architecture...
Thanks

awai54st, 2020-03-02:
Hi could you send me the error message and explain what steps you have taken to regenerate that example? Although I am no longer actively maintaining this old repo and will not add new features to it, I can try to help you with your problem via this issue thread

nehru98, 2020-03-03:
[image]
Im sorry for the bad picture quality. I actually ran lenet.tcl file from the bitstreams directory in the tcl console of vivado. The command i gave was: source lenet.tcl in vivado... Then i waited for some while and got the above errors

awai54st, 2020-03-03:
Did you run vivado hls to generate the rtl for lenet? The error message is saying that vivado cannot find the lenet rtl.  Please follow the instructions given in /hw/readme.md

nehru98, 2020-03-03:
> Did you run vivado hls to generate the rtl for lenet? The error message is saying that vivado cannot find the lenet rtl. Please follow the instructions given in /hw/readme.md

Yes, I had done the vivado hls regeneration already. Is there any other possible ways of regenerating the lenet5 design?

nehru98, 2020-03-03:
Can you send c/c++ codes for the ip blocks of lenet5, so that we can regenerate the cnn lenet project.
Thanks

awai54st, 2020-03-04:
`vivado_hls -f run_lenet.tcl` generates lenet ip in verilog using high level synthesis (HLS). The c++ source code for HLS is located under `PYNQ-Classification/hw/script_design_flow/LeNet_wrapper/`.

If you have done HLS, the it should've already generated a lenet ip (HLS-generated verilog). 

After HLS, you should run `vivado -source pynq_arch.tcl` which auto generates the vivado project for the DMA, interconnect to ARM, etc. Looking at your error message, it appears that you have successfully generated the vivado project (right?) but the project does not link to the lenet ip. Based on other users' feedback the `pynq_arch` is able to automatically link to the ip with no problem, so I need more info on how you generated the vivado project.

One thing you can also try out is to manually link the lenet ip from vivado project.

nehru98, 2020-03-04:
> `vivado_hls -f run_lenet.tcl` generates lenet ip in verilog using high level synthesis (HLS). The c++ source code for HLS is located under `PYNQ-Classification/hw/script_design_flow/LeNet_wrapper/`.
> 
> If you have done HLS, then it should've already generated a lenet ip (HLS-generated verilog).
> 
> After HLS, you should run `vivado -source pynq_arch.tcl` which auto generates the vivado project for the DMA, interconnect to ARM, etc. Looking at your error message, it appears that you have successfully generated the vivado project (right?) but the project does not link to the lenet ip. Based on other users' feedback the `pynq_arch` is able to automatically link to the ip with no problem, so I need more info on how you generated the vivado project.
> 
> One thing you can also try out is to manually link the lenet ip from vivado project.

Sure, I shall try and let you know 
Thanks

nehru98, 2020-03-10:
Hi, Could you share the design image or any snapshot of the lenet5 design you have designed and shown in: page 26 of PYNQ_CLASSIFICATION. pdf, the ip blocks in the pdf file arent clearly visible.
Thanks

closed 2020-09-14 by awai54st