awai54st/PYNQ-Classification #50: Caffe Setup:

farna-commits, 2020-04-24:
both links provided aren't working for us to download the SD card image. Also, MANUAL_INSTAL.md has a lot of issues. Can't move forward can you help?

awai54st, 2020-04-24:
Hi, I have just checked and both links work. Please provide more information on this issue.

Also, please specify the issues you found in MANUAL_INSTAL.md

farna-commits, 2020-04-24:
Okay so regarding the links they are working now fine (don't know why). 
However, the MANUAL_INSTAL.md first step in it which is: 

sudo apt-get install libprotobuf-dev libleveldb-dev libsnappy-dev libopencv-dev libhdf5-serial-dev protobuf-compiler
sudo apt-get install --no-install-recommends libboost-all-dev
sudo apt-get install libgflags-dev libgoogle-glog-dev liblmdb-dev

isn't working and tried several workarounds and still. 

Will try for sometime and will keep you posted in this thread tho. 
Thanks a lot!

farna-commits, 2020-04-24:
[image]
Hello again, here's an error in the make all step. Any suggestions?

awai54st, 2020-04-25:
I haven't been maintaining this old repo for quite a while, so it is possible that the more recent versions of these softwares have compatibility issues. That's why in the readme.md I said it is not recommended to try installing caffe on pynq yourself. You can try the sd card images that I provided instead, which already have all the dependencies installed, including caffe.

closed 2020-09-14 by awai54st