awai54st/PYNQ-Classification #53: DMA wait timed out

sabhizakaria, 2020-07-28:
Hello, First of all thank you for your amazing work and replies that helped me ( a beginner) throughout the implementation of a cnn using your framework.

I m still struggling with the "DMA wait timed out" (pictures below) whenever I try to do a hybrid implementation of lenet5 using your framework ( e.g : conv1 in fpga and the rest in CPU) 

any idea or leads that could help me please?

Thank you  

[image]

[image]

sabhizakaria, 2020-07-28:
Sorry to disturb you again, 

Basically what i would like to do is to have a conv (for example) in fpga and the rest of the lenet in cpu 
but I still have the same issue (DMA) as long as the cnn is shared between fpga and cnn 

Desperately needing your help

awai54st, 2020-07-28:
Hi, 
This error happens because the DMA at the output of your FPGA IP is waiting for the "done" signal coming out of the IP to go high. I can think of two ways that this hand-shaking could go wrong. One is that the number of inputs sent from the python script is less than what the FPGA IP is expecting, so the hardware is forever waiting for inputs before it can start executing. The other is that the FPGA IP, for some reason, never triggered the output "done" boolean signal to be 1, therefore leaving the otuput DMA waiting until time out. There could also be other possible reasons leading to this error, and having a testbench could help you debug.

sabhizakaria, 2020-07-29:
thank you for your fast reply, 
how can I see whether the ip trigger or not the boolean signal to be 1 ? 
and with a testbench , you mean that i have to debug it before hls synthesise it ? 
forgive me if my question are so dull but i have being stuck with this problem for days 

Thank you so much

awai54st, 2020-07-29:
If you look at the C++ source code of the convolutional layer's IP, you'll find an output variable "bool done". The IP is supposed to assign "1" to "done" once the IP is done processing. Apparently in your case this did not happen. So my suggestion is to use the Vivado HLS's "C++ testbench" tool and write a simulation to debug this before synthesis. This can tell you whether your FPGA IP is doing what you expect it to do. If the IP is working fine, then that probably means the problem is at the python (i.e. CPU) side, right?

sabhizakaria, 2020-07-30:
Okey i ll try the testbench tool  and get back to you 

Thank you so much

closed 2020-09-14 by awai54st