awai54st/PYNQ-Classification #54: Some confuse in config.h

alanlazzzz, 2020-07-30:
Hello, in your config.h, the bitwidth is to describe AXI transfer width, so the InputWidth is the width of input data? Are there some connections between these terminology ? Many thanks

awai54st, 2020-08-01:
Hi, Inpwidth is the precision of the matrix multiplication operands (i.e. A and B in my code), whereas bitwidth is the precision of the intermediate accumulators inside the matrix multiplication core. So bitwidth > Inpwidth because of carry bits.

alanlazzzz, 2020-08-02:
Thank you for your answer, plus, the data type of AXI_VAL and AXI_DMO_IO are all ap_int , so if I want to verify what you said in your PDF chapter 6.2.2. FIXED-POINT DATA QUANTISATION, I just need to change the data type of AXI_VAL and AXI_DMO_IO in config.h or something else? Hope for your answer, thanks again.

awai54st, 2020-08-04:
If I remember correctly those widths described in that section was talking about `Inpwidth` instead of `bitwidth`. So if you'd like to reproduce those fixed-point results you would probably want to tune `Inpwidth`. This is because `Inpwidth` determines the precision of the matrix multipliers, whereas `bitwidth` only determines the width of the data streams.

Similarly, if you'd like to try FP32 then you would probably want to change the data format of matrices `A` and `B` in `fixed_point_stream_convolution.h` and `fully_connected.h` from `ap_int` to `float`.

alanlazzzz, 2020-08-05:
Thank you so much.

alanlazzzz, 2020-08-11:
Sorry to distrub you again, I change the data format of matrices A and B in fixed_point_stream_convolution.h and fully_connected.h to ap_fixed, and the inpwidth in config.h, then I repacked the ip core and generate the bitstream, modify the bitstram name in conv_fpga.py, but the accuracy in Using a Caffe Pretrained Network - CIFAR10.ipynb is always 0.728(it should be smaller than this), are there any trouble in my options or I need to do more options? Thanks again.

awai54st, 2020-08-14:
Have you updated `quant_scale` after you changed the inpwidth? Otherwise there might've been overflow/underflow.

alanlazzzz, 2020-09-26:
In your config.h, the type of data in AXI_VAL is ap_int, so the data tansfer in the network is all ap_int? But some data like weight in caffemodel is not just ap_int(like -1.231424), I got some confuse about this, and in your original code, the type of matrix in fc and conv is all ap_int, I mean if I want to store the weight just like 1.58418941, it could be cut to 1 or other number?(I know it cannot to be this, because the result of network`s precision is very high.) Many thanks, hoping for your answer.

awai54st, 2020-10-03:
Hi, please see how `quant_scale` is applied in each layer. Although the data format is AXI_VAL, `quant_scale` will scale the outputs of each layer such that there is only one bit to the left of the radix point, and the remaining bits will represent the fractional part of the data.

alanlazzzz, 2020-10-14:
Thanks a lot!

closed 2020-10-14 by alanlazzzz