awai54st/PYNQ-Classification #55: 4.完整工程与源码（百度网盘链接）

ciky-joh, 2020-09-14:
4.完整工程与源码（百度网盘链接）
方便分享么  大佬  如果方便加q哈 3045696011

awai54st, 2020-09-14:
您好，工程源码全在这个repo里。PYNQ映像请看主页链接（已测有效）

ciky-joh, 2020-09-14:
感谢哈  大佬   刚才没注意看 我是新手  我还想求教下   我现在有一个zynq-7020 开发板  然后有PYNQ镜像  我想把您这个移植到我的开发板上  但是不知道怎么做  您能详细说下大概需要几个步骤么  我的项目想用pytorch框架搭建的网络然后调用您这个overlay 对我的网络进行加速  这个能实现么   如果方便加q哈   万分感谢

awai54st, 2020-09-14:
您可以直接下载我准备的映像，里面有PYNQ OS+项目代码可以直接用Jupyter Notebook跑，和树莓派差不多。这个项目比较早，当时训练我用的是Theano。换framework理论上肯定是可行的，因为这个项目是inference比较简单。不过换网络的话要同时修改硬件IP（Vivado HLS，C语言）和ARM的数据传输代码（Python）。具体如何做可以参照repo里的closed issues我的回答

ciky-joh, 2020-09-15:
非常感谢您的回答哈  我的开发板是正点原子版的ZYNQ-7020这个上面有PYNQ镜像-v2.5的 ，这个应该不影响吧？还有一个问题，这个开发板算力是多少呢？我想做图像分割，我的模型在电脑上测试是15Gflops左右，这个能部署到7020上么？能不能达到实时性，您有没有参考的算力值呢？

awai54st, 2020-09-15:
开发板应该和我当时用的差不多，镜像应该可以兼容。我的镜像是早期的PYNQ OS版本，好像是v1.x。就像树莓派一样，在使用的时候把镜像刷到SD卡上需要将卡格式化。建议原始镜像留一个备份。FPGA的算力很难估算吧，因为它会根据IP的scheduling，计算精度，甚至synthesis tool的mapping效率等等因素发生较大改变。假设使用与本程序相同的dataflow paradigm，可以参考我之前[这篇文章](https://arxiv.org/abs/1901.06955)的roofline model analysis。文中分析了ZC706，不过你可以尝试用相同方法分析xc7z080的roofline。15Gflops 是fp32, 而本项目用的是INT8或INT16。假设15Gops使用INT8，那么ZC706理论可行，xc7z080不好说，估计悬

ciky-joh, 2020-09-16:
大佬  非常感谢您的细心回答哈  祝您笑口常开哈 我的这部分不太懂“修改硬件IP（Vivado HLS，C语言）和ARM的数据传输代码（Python）”不知道具体修改那个文件呢？我先研究研究，有问题再来问你哈   给您太麻烦了 再次感谢哈

closed 2021-05-09 by awai54st