awai54st/PYNQ-Classification #59: xc7z020镜像问题

wminghao712, 2020-11-10:
您好，EW，您的工程非常有学习得意义，我这两天学习了一下您得工程，我现在有一块XC7z020-clg400-2的开发板，但是我用您准备好的镜像下载到板子上时，发现无法启动系统的问题，就是PUTTY是黑屏状态，我不确定是不是镜像无法适用于XC7Z020-clg400-2,所以希望得到您的帮助，如果确实不适用，我应该在XC7Z020-clg400-2的ubuntu镜像中安装caffe等库是吗？期待您的回复
EN：
Hello, EW, your project is very meaningful to learn. I have studied your project in the past two days. I now have a development board of XC7z020-clg400-2, but I will download it to the board with the image you prepared When I found that the system could not be started, the PUTTY was in a black screen state. I am not sure if the mirroring is not applicable to XC7Z020-clg400-2, so I hope to get your help. If it does not apply, I should be in XC7Z020-clg400-2 Is caffe and other libraries installed in the ubuntu mirror? Looking forward to your reply

awai54st, 2020-11-22:
您好，不好意思回复得有些晚。我不是很确定这个项目是否能完全兼容您的开发板，不过我认为PUTTY是否黑屏和FPGA版本无关，而是和ARM CPU有关，而且我感觉CPU这边不应该出现兼容问题。请问您是否尝试在PUTTY出现黑屏的时候ctrl C或者回车？我隐约记得当年我使用USB serial端口接PUTTY的时候需要ctrl C才能出现terminal

SaeedHassan243, 2021-01-10:
I have the same issue..

awai54st, 2021-01-11:
Hi SaeedHassan243, I haven't used PYNQ in a while, but I wonder if you have tried pressing enter or ctrl C when you see the blank terminal? [This post](https://pynq.readthedocs.io/en/v2.5.1/getting_started/terminal.html) says that you need to press enter first to see the terminal.

closed 2021-05-09 by awai54st