awai54st/PYNQ-Classification #65: the block design created in vivado is different from the block design in your video

bahadir25, 2021-04-12:
Hello, I did script design but the block design created in vivado is different from the block design in your video. Am I doing something wrong?

[image]

and 

[image]

awai54st, 2021-04-12:
Hi I do not see any problem in your screenshots. There is one convolutional layer followed by a maxpooling layer. Could you please specify the differences that you refer to?

bahadir25, 2021-04-13:
Sorry, I'm trying to learn some things by doing this project. Are they the same thing? I'm asking because I don't know.

bahadir25, 2021-04-13:
Please tell me which of the IPs in the bottom picture is convolution layer, pooling layer, fully connected layer.

awai54st, 2021-04-13:
Hey no problem at all, glad that this old project can be helpful to you. 

So with script flow I put the entire network into a single HLS IP cores (all IPs with the "HLS" logo are IPs generated from HLS). In your bottom picture, the "cifar_10_0" contains the entire network. The other three HLS IPs ("simple_sum" etc) are not relevant to this project and they were for debugging purpose only.

A bit of history with this project: when I did the original version for my master thesis, there is only the graphical flow as you have seen in the video. I thought the drag and drop feature can make the development process look interactive and intuitive. However, based on feedback it turns out that exposing the network in graphical GUI leads to loads of issues in different dev environments (different OS, different Vivado version, etc). So in the end I provided the script flow, which integrates the whole network into one single HLS IP. The network can be configured with C++ templates. This is much more robust.

bahadir25, 2021-04-14:
Understood. Thank you very much for your detailed explanation. What is the detailed structure of the network? So how many convolution layers, pooling layers, fully connected layers are there? That's why I'm asking, as you can see in the picture below, there are 3 from each layer. You use one of them all in grafical design.

bahadir25, 2021-04-14:
[image]

bahadir25, 2021-04-14:
[image]

awai54st, 2021-04-14:
For script flow you can look at PYNQ-Classification/hw/script_design_flow/CIFAR_10_wrapper/CIFAR_10_wrapper.cpp

For the CIFAR-10 network, see line [31](https://github.com/awai54st/PYNQ-Classification/blob/1135961dcdb21ba57aeefe132f1fe95d5b1e25ee/hw/script_design_flow/CIFAR_10_wrapper/CIFAR_10_wrapper.cpp#L31) to 43.

closed 2021-05-09 by awai54st