awai54st/PYNQ-Classification #66: 自定义数据集

Daniel-bboy, 2021-04-16:
您好，我想问一下，您这个是做的数据集时cifar-10是吗？那如果要自己换成自定义的数据集的话，应该怎么做呀？

awai54st, 2021-04-16:
您好，我当时做了两个数据集，CIFAR-10和MNIST (LeNet)。如果是自定义数据集的话，我猜测网络结构也要换吧？

换数据集应该不难。用新数据集训练之后将训练好的weights (.h5)替换这里之前的weights，假如网络结构不换的话那么就可以直接用现有的IP做inference了。

假如换网络结构的话，会复杂一些。ARM方面软件部分需要新网络结构的信息(.prototxt)，并且在ARM内存 (DRAM)中存入与读取正确的矩阵大小(conv_fpga.py)，最后在FPGA硬件方面要改变网络结构，重新synthesis硬件IP，替换bitstream。软件与硬件的输入输出多少数据一定要对的上，不然DMA会在从DRAM读写时lock住。

Daniel-bboy, 2021-04-17:
嗯嗯，谢谢您的回答，的确就是一个问题就是是否可以直接在您目前生成的yolov2的ip上直接更换数据集来完成。因为按道理来说，如果只是更换数据集的话，对网络本身其实是没有影响的吧？
然后，因为最近在做这个相关的项目，想问问是否方便加个联系方式，可以和您讨论，谢谢您！我的微信是18282233378。

-----------------------------------

Daniel-bboy, 2021-04-28:
您好，我想问一下，.h5文件需要怎么生成？那个是自己训练的caffemodel自动生成的吗？谢谢

StephQAZ, 2021-07-06:
您好，我想问一下，我修改了LeNet_wrapper.cpp中的网络结构，并重新生成了ip核。但我不知道我需要修改conv_fpga.py中的哪些内容，请问我只需要修改以下图片中的划横线的内容吗，还是需要修改其他地方？谢谢！
[image]

awai54st, 2021-07-06:
嗯嗯这些确实都可能需要改，还有那个18是为了实现quantisation而做的scaling，是我试出来的，所以也可能需要更改。。。

别的地方有可能还有，不过时间久了我实在记不清了。

StephQAZ, 2021-07-09:
> 嗯嗯这些确实都可能需要改，还有那个18是为了实现quantisation而做的scaling，是我试出来的，所以也可能需要更改。。。
> 
> 别的地方有可能还有，不过时间久了我实在记不清了。

谢谢您的回复。我还是放弃了用HLS来进行CNN加速。这几天我尝试采用DPU-PYNQ + Vitis-AI + ultra96 进行神经网络加速，目前效果还不错。

closed 2021-07-27 by awai54st