awai54st/PYNQ-Classification #67: make compile_graphical failed

CHen-Hua-Luen, 2021-04-22:
您好，我在HLS Command Prompt 输入make compile_graphical后，Vivado自动打开，并且有Error，想起问您该怎么处理? 是否有可能因为我是在Windows下执行? 若改用Linux的话有可能解决吗? 谢谢您
[image]
[image]
[image]
[image]

awai54st, 2021-04-27:
您好，graphical design flow有很多用户反馈说不稳定。所以我建议你使用script flow来设计。

CHen-Hua-Luen, 2021-05-01:
谢谢

closed 2021-05-01 by CHen-Hua-Luen