awai54st/PYNQ-Classification #69: 有关Python code的问题

CHen-Hua-Luen, 2021-05-01:
您好，想请问

1.FPGA Deployment (CIFAR10 Layer)
   与FPGA Deployment(QuickTest Function)的差异是什么?

2.其中的OFMCH是指要分类的个数吗?

谢谢

CHen-Hua-Luen, 2021-05-04:
3.请问资料预处理一定要为零均质化吗?若直接在Jupyter中改成自己的预处理方式会有问题吗?

CHen-Hua-Luen, 2021-05-04:
4.请问为何只有把权重送到FPGA,却没有送Bias?

awai54st, 2021-05-09:
1. 第一个是将FPGA IP的调用嵌套进一个Theano的custom layer。但是实际运行较慢。第二个是将其嵌套进一个python function，运行较快
2. Output Feature Map CHannel number
3. 具体要看预处理方式吧
4. 因为这个model训练时没有用bias

CHen-Hua-Luen, 2021-05-18:
我把资料集改成自己的，而架构和分类数目不变，但是送进去都被分类为类别1，请问你有遇过这样的问题吗，或知道可能是什么样的原因会造成这个问题?谢谢

CHen-Hua-Luen, 2021-05-24:
我把数据集换成自己的，且是四种分类，但FPGA却还是分类了十种类别，请问我要改分类数量的话，是除了Using a Caffe Pretrained Network里面的架构的最后一层要改跟conv_fpga内的FPGA_CIFAR10的input_val = np.append([0, batch_size, 5, 3, 32, 10, 1, 2], input_val) 改为input_val = np.append([0, batch_size, 5, 3, 32, 4 , 1, 2], input_val)，还有哪里的程式要改吗? 谢谢

awai54st, 2021-05-24:
如果想将最后一层网络的输出由10改为4的话，则必须改FPGA IP,重新综合，生成并替换新的bitstream。不然FPGA的DMA数据差6个未流出，就会锁死

CHen-Hua-Luen, 2021-05-25:
不同的资料集但都是十个类别，用caffe训练 dump权重没事,但用keras训练并dump权重却会发生DMA wait timed out, 请问这可能的原因是什么?

CHen-Hua-Luen, 2021-05-27:
3.我如果用caffe训练，而bitsream档自己重新产生，但架构不变，没有动任何东西，但使用这.bit档却会发生DMA waited timeout, 而直接用你映像档里面.bit的则不会，请问这可能的原因是什么?

awai54st, 2021-05-30:
请问weight的数量是否也发生变化了呢？在向FPGA IP输入weight的时候如果weight的尺寸对不上的话，那么一样会发生timeout

CHen-Hua-Luen, 2021-06-02:
解决了谢谢 想再请问

1.CIFAR10 Layer 跟 QuickTest Function 得到的准确度差很多是正常的吗? EX:0.09, 0.925

2.data_zeromean*64 为何要乘 64?

谢谢

awai54st, 2021-06-06:
1. 如果我没记错的话应该是一样的
2. 在使用FPGA进行inference的时候有量化（可选8或16比特，需要改变[这一行](https://github.com/awai54st/PYNQ-Classification/blob/1135961dcdb21ba57aeefe132f1fe95d5b1e25ee/hw/script_design_flow/CIFAR_10_wrapper/config.h#L6)的scaling factor重新综合）
3. 我没记错的话这里和8比特量化的格式有关。好像是6位整数2位小数？这个可以自己定义自己调整

CHen-Hua-Luen, 2021-06-08:
解决了，是我不小心改到程序了，谢谢

closed 2021-06-08 by CHen-Hua-Luen