awai54st/PYNQ-Classification #73: lenet graphical design file

mustafatasci42, 2021-06-13:
i want to study lenet graphical design files how can i access these files.

awai54st, 2021-06-19:
Hi. What design files do you refer to? For the C++ HLS source files please see [here](https://github.com/awai54st/PYNQ-Classification/tree/master/hw/graphic_design_flow/CIFAR-10). For the tcl script that generates the vivado IP block containing the network, please see [here](https://github.com/awai54st/PYNQ-Classification/tree/master/hw/graphic_design_flow/CIFAR-10/SKETCHPAD). The generation of IPs has been automated by the Makefile (as explained in [here](https://github.com/awai54st/PYNQ-Classification/blob/master/hw/README.md)).

Please be adviced that the graphical design flow, as suggested by many users' feedback, is less stable than the script design flow. Please use the latter instead.

mustafatasci42, 2021-06-20:
thanks for your reply. I rebuilt the rtl files

closed 2021-11-30 by mustafatasci42