awai54st/PYNQ-Classification #77: Error rebuilding from hw/readme using tcl files.

shobhitt10, 2022-01-18:
Hi, I am trying to rebuild the project. So, while following the hw/readme 
Running the command **vivado_hls -f run_cifar_10.tcl** gave following error: 

****** Vivado v2016.2 (64-bit)
  **** SW Build 1577090 on Thu Jun  2 16:32:40 MDT 2016
  **** IP Build 1577682 on Fri Jun  3 12:00:54 MDT 2016
    ** Copyright 1986-2016 Xilinx, Inc. All Rights Reserved.

source run_ippack.tcl -notrace
bad lexical cast: source type value could not be interpreted as target
    while executing
"rdi::set_property core_revision 2201181623 {component component_1}"
    invoked from within
"set_property core_revision $Revision $core"
    (file "run_ippack.tcl" line 755)
INFO: [Common 17-206] Exiting Vivado at Tue Jan 18 16:23:10 2022...
CRITICAL WARNING: [IMPL 213-28] Failed to generate IP.
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [HLS 200-112] Total elapsed time: 87.125 seconds; peak memory usage: 388 MB.

Can anyone help.

mustafatasci42, 2022-01-26:
This issue is due to the system date. Temporarily change your pc date to 2021 this problem will be solved.

shobhitt10, 2022-01-27:
Thankyou so much. Issue got resolved after changing date