awai54st/PYNQ-Classification #80: cannot make cnn block design

shobhitt10, 2022-01-28:
Following the hw/readme, I could rebuild the overall block design with cifar10 ip as shown below.
[image]

But, **I want to alter the structure inside the cifar10 ip** i.e. change the no. of layers and other parameter similar to what is shown in the youtube tutorial.
I could not find where is the block design of the cifar 10 ip. 
Can anyone help in this.

PS: My final goal is to use some modified or the same cnn model for classification on some other dataset.

Remel-cd, 2022-02-07:
hellow, I have a problem when export IP in HLS.
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
I want to know you rebuild the hls project in win or ubuntu？

shobhitt10, 2022-02-07:
I build in windows

Remel-cd, 2022-02-07:
aha, I changed the date to 2021 it works
tks